ros-drivers/mocap_optitrack #12: Full NatNet support with Timestamps

jwag, 2015-01-12:
I'm not sure that this is the right place to do this, but I didn't know where else to start this. I am interested in getting the timestamp information from the NatNet Packets and republishing it over ROS. It looks as though it would be straight forward to pull the timestamp out the packet, but I wasn't sure what the best way would be to republish that time. I would be willing to code this up, but I thought I would ask how it should be done before I start.

Thanks

mikepurvis, 2015-01-12:
The PoseStamped which is published is already stamped with ROS (system) time, so you just need to publish the relationship between ROS time and NatNet time, right?

If so, there's a specific message for this purpose, presently used to relate ROS time to GPS time: http://docs.ros.org/indigo/api/sensor_msgs/html/msg/TimeReference.html

Though, why is it that they are different? Can you not have the Tracking Tools and ROS computers both synced pretty accurately to UTC?

jwag, 2015-01-12:
Great! I didn't know this existed.

The reason that I want the time from the NatNet packets is so that I can synchronize the data from the optitrack with another system accurately. The time that this node receives the data is not the time the frame was actually captured. 

Another problem is that the timestamp is a value determined by the local camera/optihub clocks and not the host computer. Synching up the system accurately is likely going to require an external device time synchronized with the host machine and listening to the sync out to timestamp frame captures. But this local time info may end up helping though.

mikepurvis, 2015-01-12:
Oh, ugh, I didn't realise that the PoseStamped timestamps were not the original source timestamps— those should definitely be stamped with the time that OptiTrack says the pose is from, not the arbitrary moment which it arrived at the ROS computer.

jwag, 2015-01-12:
That was kinda what I was thinking too, but it will be difficult to do that because the optitrack system doesn't have that capability built in. I have been conversing with their engineers about this, and they have basically told me I will have to figure out the timing thing on my own. It is pretty surprising to me that they don't actually have accurate timestamps in terms of the host computers clock. They do provide a method for synchronizing frames with cameras using SMPTE timecodes, but the accuracy is only around 40ms. And it only works for ethernet systems that have an ESync and some sort of SMPTE timecode generating device plugged into it.

tonybaltovski, 2015-01-13:
@mikepurvis, sorry for not specifying that the timestamps are when the message arrives on the ROS computer.  Like @jwag, I was unable to find a good solution for synchronizing time between both systems.  There is a time stamp from Motive that we can decode too make it closer to the real time but still not accurate.  We have the USB cameras and would have to find a way to sync the hubs using the SMPTE external out coaxial cable and host computer.   Ultimately, I believe using just a Pose message is not very practical since most applications use a PoseStamped because the header information is needed (for the frames and change in time).  However, a better time synchronizing solution is still needed.

TZVIh, 2019-05-06:
Hi, 
Do you have solution for the time synchronizing problem?

closed 2021-02-25 by tonybaltovski