ros-drivers/mocap_optitrack #22: No timestamp when using mocap_optitrack

pchatrath, 2016-07-30:
I use mocap_optitrack to get data from Motive to ROS. I implemented the complete software pipeline till my PX4 uav. I get mocap timeout and xy timeout error. I figured out locap position estimator in px4 requires difference between 2 timestamps to be less than 0.2s. However when I check the topic published using mocap_optitrack I dont see any timestamp in the header. What am I missing. Why isnt there any timestamp in the topic? Any ideas?

jhabhargav26, 2016-07-31:
I am trying to get the same. The natnet documentation says that the packet includes the timestamp from Motive but I don't know how to decode it. Any leads would be highly appreciated. Thanks

tonybaltovski, 2016-08-04:
See https://github.com/ros-drivers/mocap_optitrack/issues/12 for a previous discussion on this.

closed 2017-02-15 by tonybaltovski