ros-drivers/mocap_optitrack #23: track multiple rigid bodies

Akash-Pardasani, 2016-12-10:
@tonybaltovski Could you please guide on how to track multiple rigid bodies using mocap_optitrack.
It works fine with a single body but when tracking two, i get the following output on terminal 
[image]
I am using motive 1.10.1 and I also found this online(it doesn't work as well)[multiple_rigid_bodies](http://answers.ros.org/question/217341/multiple-rigid-bodies-crash-optitrack-node/)
Any help would be appreciated

tonybaltovski, 2017-01-08:
I use this fine with Motive 1.7.0.  Can you provide your configuration for this node and Motive?  Also, try launch with the screen option to get more information.

lericson, 2017-02-15:
I had the same issue. Fixed in #24 .

neudinglei, 2017-05-03:
@Akash-Pardasani @tonybaltovski @lericson @wjwwood @trainman419 
Now,I want to do collaborative research with my colleagues by the Crazyflie2.0. I am using the optitrack system and ros package mocap_optitrack.The problem is that it can not communicate with three or more Crazyflie2.0。Can you give me some advice and solutions?Thank you very much!

closed 2021-02-25 by tonybaltovski