ros-drivers/mocap_optitrack #26: Failed to set socket option: EINVAL

m1577, 2017-07-12:
Hello! I want use this package.

I have a windows pc with Motive 1.10 and ubuntu pc with ROS kinetic.

I used git clone the https://github.com/ros-drivers/mocap_optitrack repository in my catkin workspace , then I compile my wokspace usin catkin_make.

I modified the config.yaml file, I change the Rigid body name and multicast adress.

I run launch file: roslaunch mocap_optitrack mocap.launch

But I am getting a error.

Do I have to use other optitrack motive version? I am using motive 1.10
What ip adress do I have to use? I am using a windows pc and ubuntu pc with connected via LAN. so, Do I have to use windows ip adress or ubuntu ip adress?

Regards!!

yzeleke, 2018-03-03:
I have same exact problem

yzeleke, 2018-03-03:
Hello guys the reason for getting what():  Failed to set socket option: EINVAL error is use of invalid multicast address. I found a type on the http://wiki.ros.org/mocap_optitrack page. They mentioned the multicast address should be the ip of the machine running rosnode. This is WRONG. 
Multi cast address should be in the range 
224.0.0.0 through 239.255.255.255.

The Fix:
On motive change the multicast interface to any number between that range eg. 230.0.0.1 and also change the mocap.yaml file inside mocap_optitrack-> config to 
optitrack_config:
        multicast_address: 239.255.42.99

closed 2021-02-25 by tonybaltovski