ros-drivers/mocap_optitrack #29: Multiple rigid bodies problem

ZhiangChen, 2017-09-17:
When using multiple rigid bodies, it has a problem.

Solution:
In the function void MoCapDataFormat::parse() of file mocap_datapackets.cpp.
Add seek(2) after line154:
seek(sizeof(float)); 
seek(2);

guozhenglong, 2017-09-25:
Really Thanks! This solved my problem perfectly!

Vibashan, 2018-05-29:
process[mocap_node-1]: started with pid [6627]
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
================================================================================REQUIRED process [mocap_node-1] has died!
process has died [pid 6627, exit code -6, cmd /home/nikes/catkin_ws/devel/lib/mocap_optitrack/mocap_node __name:=mocap_node __log:=/home/nikes/.ros/log/2ac5a4a0-63a0-11e8-88d4-30e37a02f7e7/mocap_node-1.log].
log file: /home/nikes/.ros/log/2ac5a4a0-63a0-11e8-88d4-30e37a02f7e7/mocap_node-1*.log
Initiating shutdown!
================================================================================
[mocap_node-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

I added seek(2);..... but still not working the same error appears again, even i tried out seek(3);...etc..

Vibashan, 2018-09-29:
The general solution is to add seek(sizeof(long long)); and inorder to track 'n' rigid bodies  in mocap.yaml file u have to add n times
'n':
        pose: Robot_n/pose
        pose2d: Robot_n/ground_pose
        child_frame_id: Robot_n/base_link
        parent_frame_id: world

closed 2021-02-25 by tonybaltovski