ros-drivers/mocap_optitrack #33: Optitrack and Crazyflie communication

s3xyxavy, 2018-02-11:
Hi, 

I am currently working on a project with in having a communication with mocap and crazyflie. I realized that when I was checking the quarternions on RVIZ, the quarternion for rotation in y plane is inverted is there any way I could change the parameter to make the y result inverted ( + / - ) sign. TIA =)

Warm Regards,
Xavier

tonybaltovski, 2018-02-11:
What version of Motive or Tracking tools are you using?

s3xyxavy, 2018-02-12:
Hi Tony,
 
I am using 1.7 On Motive, there isn’t toggle for y-up on motive for this version. 

Warm regards,
Xavier

s3xyxavy, 2018-02-14:
Hi Tony, where do I look at the new-and-old-support-updated branch?

tonybaltovski, 2018-02-14:
https://github.com/ros-drivers/mocap_optitrack/tree/new-and-old-support-updated

closed 2018-04-02 by s3xyxavy