ros-drivers/mocap_optitrack #38: Updating for New ROS Builds

magicbycalvin, 2018-08-07:
Would it be possible to add this to the new ROS builds so that people do not have to build it from source?

Thanks for all your hard work!

tonybaltovski, 2018-08-08:
@magicbycalvin I have limit access to a working system.  Are you currently using this?  If so, which branch are you using?

magicbycalvin, 2018-08-08:
@tonybaltovski I am still getting everything up and running but I went with the master branch (which I am just now realizing is 3 commits older than the new-and-old-support-updated branch). What would it take to be able to install it using apt-get with some of the newer ROS builds like kinetic, lunar, and melodic?

Thanks!

tonybaltovski, 2018-08-08:
@magicbycalvin, just for me to release it but my only reservation was to add support for the changes in Motive.

tonybaltovski, 2018-08-14:
@magicbycalvin does the `new-and-old-support-updated` work for your setup?  If it does, I will do a release.

plusk01, 2019-02-27:
@tonybaltovski what is the status of making a release and getting this into the build farm?

FYI, `master` is working with my setup:

```
[ INFO] [1551307844.790725441]: NATNet Version : 3.1.0.0
[ INFO] [1551307844.790736189]: Server Version : 2.1.0.0
```

tonybaltovski, 2021-02-25:
This got released. Closing due to inactivity.  Please feel free to re-open if issue persists.

closed 2021-02-25 by tonybaltovski