ros-drivers/mocap_optitrack #49: How can I use the ROS2 port with Dashing?

jeferrandiz98, 2020-10-30:
Hello, I would like to use mocap_optitrack on ROS2 dashing. I downloaded and compiled the crystal-devel version. However, I am having some issues, I think it is something related to parameters.
The node can not read the parameters correctly and it can not load the ID's of the rigid body.

Is there any change I can make to use this package on Dashing, or is there any other package to use read from optitrack net on ROS2?

Thank you so much for your help in advance.

Raphtor, 2021-07-19:
I was able to get it to compile and run on ROS2 Dashing  by changing the node declaration in `mocap_node.cpp`

```
  rclcpp::NodeOptions ops = rclcpp::NodeOptions();
  ops.allow_undeclared_parameters(true);
  ops.automatically_declare_parameters_from_overrides(true);
  rclcpp::Node::SharedPtr rclcpp_node = std::make_shared<rclcpp::Node>("mocap_node",ops);
```
and also removing any dependence on `tf` in `rigid_body_publisher.{cpp,h}` (I think there's something wrong here but I didn't need TF anyway). However, I am having difficulty getting the node to actually publish anything.

tonybaltovski, 2022-01-18:
Closing due to inactivity.

closed 2022-01-18 by tonybaltovski