ros-drivers/mocap_optitrack #59: Connection issues between ROS Melodic and Optitrack Streaming Engine

geothan, 2021-03-19:
I can't retrieve the pose of a 3D object in ROS . I have 2 computers, one with Windows 10 and one with Ubuntu 18. 

The Windows PC has Motive: Tracker 2.2.0 with the following settings in Optitrack Streaming Engine:
1. Bradcase Frame Data: On
2. Local Interface: the computer's IP, not loopback
3. Rigid Bodies: On
4. Transmission Type: Multicast
5. Command Port: 1510
6. Data Port: 1511
7. Multicast Interface: 239.255.42.99

 The Linux PC has ROS Melodic and with following installation steps:
`cd ~/catkin_ws/src`
`git clone https://github.com/ros-drivers/mocap_optitrack.git`
`cd ..`
`catkin build`

The project was built with the below modifications:
1. Command Port: 1510
2. Data Port: 1511
3. Multicast Interface: 239.255.42.99

I run the following command :
`roslaunch mocap_optitrack mocap.launch`

and get the below:

roslaunch mocap_optitrack mocap.launch
... logging to /home/george/.ros/log/0bbc8108-88c0-11eb-a872-000c2978f07e/roslaunch-ubuntu-69992.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:38971/

SUMMARY
========

PARAMETERS
 * /mocap_node/optitrack_config/command_port: 1510
 * /mocap_node/optitrack_config/data_port: 1511
 * /mocap_node/optitrack_config/enable_optitrack: True
 * /mocap_node/optitrack_config/multicast_address: 239.255.42.99
 * /mocap_node/rigid_bodies/1/child_frame_id: Robot_1/base_link
 * /mocap_node/rigid_bodies/1/odom: Robot_1/Odom
 * /mocap_node/rigid_bodies/1/parent_frame_id: world
 * /mocap_node/rigid_bodies/1/pose2d: Robot_1/ground_pose
 * /mocap_node/rigid_bodies/1/pose: Robot_1/pose
 * /mocap_node/rigid_bodies/1/tf: tf
 * /mocap_node/rigid_bodies/2/child_frame_id: Robot_2/base_link
 * /mocap_node/rigid_bodies/2/odom: Robot_2/Odom
 * /mocap_node/rigid_bodies/2/parent_frame_id: world
 * /mocap_node/rigid_bodies/2/pose2d: Robot_2/ground_pose
 * /mocap_node/rigid_bodies/2/pose: Robot_2/pose
 * /mocap_node/rigid_bodies/2/tf: tf
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    mocap_node (mocap_optitrack/mocap_node)

auto-starting new master
process[master]: started with pid [70002]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 0bbc8108-88c0-11eb-a872-000c2978f07e
process[rosout-1]: started with pid [70013]
started core service [/rosout]
process[mocap_node-2]: started with pid [70020]
[ WARN] [1616164398.362593479]: Could not get server version, using auto

jadhm, 2021-03-22:
@geothan  Hi, did you define the the id streaming of the rigidbody ? because what you see in the config yaml file of this package that there are 2 rgigbodies are defines the first one with the streaming id 1 and the other is 2 so please make sure that the rigigbody in motive has the streaming id of 1 or 2 

[image]

geothan, 2021-03-24:
Hi @jadhm!

i think that i i've defined properly the streaming ID. As for the second object, it doen't exist. I left the rigidbodies of mocap.yaml as they are written.Could the specific driver not be compatibe with motive 2.2?
[image]

jadhm, 2021-03-24:
Hi @geothan  
That's weird but it should work because I am also using motive 2.2.0 and the driver is compatible with it. could you please try to launch it using --screen so I can see the initialization logs 
 `roslaunch mocap_optitrack mocap.launch --screen` 

 and if you try `rostopic list` would you see the all the topics ?

geothan, 2021-03-24:
Hi @jadhm ,

Here is a screenshot of launch file execution:

[image]

The result of topics is the following:

[image]

The first time that i tried the above settings, topics of robot1 weren't visible only parameter_descriptions and parameter_updates topics. These topics are empty, is that normal?

[image]

Finally, when i run 
`roslaunch mocap_optitrack mocap.launch & rosrun rviz rviz`

Shall i see my robot in rviz? Because i an empty world.

[image]

jadhm, 2021-03-25:
It seems it works just fine ! I see that you can read the topic `/mocap_node/Robot_1/pose` which is basically the position of the tracked object. So I don't think you have any problems and for visualizing the pose in RVIZ you should add the topic manually and also configure the Fixed frame.

lixryjz, 2021-12-18:
> 
hi，yet I am very happy to see that, but i have down the fixed frame and add topic as you said in RVIZ,there is only a red line with an arrow.can you tell me how to do?i hope you can screenshot

chennuo0125-HIT, 2023-03-17:
@geothan @lixryjz the problem has been solved ? i get same problem as follows:
```
[ WARN] [1679050884.023240442]: Could not get server version, using auto
```
and no more print