ros-drivers/mocap_optitrack #73: Cannot Stream Data / Motive 3.0.0 / ROS Noetic

gsethi2409, 2022-06-02:
I am trying to stream data into ROS Noetic [Ubuntu 20.04 on Dell XPS] via Motive 3.0.0 [Windows 10 on a Dell System].

Here are my Motive settings. The Local Interface is set to the Motive system's IPv4-
[image]

Config on the ROS end is as follows-

```
rigid_bodies:
    '1':
        pose: Robot_1/pose
        pose2d: Robot_1/ground_pose
        odom: Robot_1/Odom
        tf: tf
        child_frame_id: Robot_1/base_link
        parent_frame_id: world
optitrack_config:
        multicast_address: 239.255.42.99
        command_port: 1510
        data_port: 1511
        enable_optitrack: true
```

When I run `roslaunch mocap_optitrack mocap.launch & rosrun rviz rviz`, here are my logs:

```
[2] 94648
[ INFO] [1654133406.946243021]: rviz version 1.14.14
[ INFO] [1654133406.946390637]: compiled against Qt version 5.12.8
[ INFO] [1654133406.946399913]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1654133406.951958820]: Forcing OpenGl version 0.
... logging to /home/gsethi2409/.ros/log/2f08d2ae-e212-11ec-8a56-abda5e723894/roslaunch-gsethi2409-94648.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://gsethi2409:42521/

SUMMARY
========

PARAMETERS
 * /mocap_node/optitrack_config/command_port: 1510
 * /mocap_node/optitrack_config/data_port: 1511
 * /mocap_node/optitrack_config/enable_optitrack: True
 * /mocap_node/optitrack_config/multicast_address: 239.255.42.99
 * /mocap_node/rigid_bodies/1/child_frame_id: Robot_1/base_link
 * /mocap_node/rigid_bodies/1/odom: Robot_1/Odom
 * /mocap_node/rigid_bodies/1/parent_frame_id: world
 * /mocap_node/rigid_bodies/1/pose2d: Robot_1/ground_pose
 * /mocap_node/rigid_bodies/1/pose: Robot_1/pose
 * /mocap_node/rigid_bodies/1/tf: tf
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    mocap_node (mocap_optitrack/mocap_node)

ROS_MASTER_URI=http://localhost:11311

[ INFO] [1654133407.262699054]: Stereo is NOT SUPPORTED
[ INFO] [1654133407.262763183]: OpenGL device: NVIDIA GeForce RTX 3050 Ti Laptop GPU/PCIe/SSE2
[ INFO] [1654133407.262775517]: OpenGl version: 4.6 (GLSL 4.6).
process[mocap_node-1]: started with pid [94716]
[ WARN] [1654133407.295136451]: Could not get server version, using auto
```
And `rostopic list` gives me:

```
rostopic list
/mocap_node/optitrack_config/parameter_descriptions
/mocap_node/optitrack_config/parameter_updates
/rosout
/rosout_agg
```

I have been stuck on this for a really long time. I cannot seem to figure out the issue here! Is this a compatibility issue? Please help!

HazemElrefaei, 2023-08-08:
I have the same problem

Murad275, 2023-08-17:
Did someone figure out this yet? @HazemElrefaei @mengchaoheng @gsethi2409

HazemElrefaei, 2023-08-18:
@Murad275, I used another Repo and it worked for me: https://github.com/L2S-lab/natnet_ros_cpp.git