ros-drivers/mocap_optitrack #75: No server info request, parsing data message aborted

ljaniec, 2022-07-14:
Hello,

I tried to build this package and connect with it Motive 1.7.2 (64-bit) in the laboratory and the ROS 2 Galactic node. 
I got this output (the same was in the log files).

```
ros2@mrlab-server:~/mocap_ws$ ros2 launch mocap_optitrack mocap.launch.py 
[INFO] [launch]: All log files can be found below /home/ros2/.ros/log/2022-07-14-16-49-35-680168-mrlab-server-126848
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mocap_node-1]: process started with pid [126850]
[mocap_node-1] [WARN] [1657810175.838541182] [mocap_node]: Could not get server version, using auto
[mocap_node-1] [INFO] [1657810175.838652649] [mocap_node]: Creating socket...
[mocap_node-1] [INFO] [1657810175.838673028] [mocap_node]: Setting socket options...
[mocap_node-1] [INFO] [1657810175.838683314] [mocap_node]: Local address: 0.0.0.0:1511
[mocap_node-1] [INFO] [1657810175.838690114] [mocap_node]: Binding socket to local address...
[mocap_node-1] [INFO] [1657810175.838698600] [mocap_node]: Joining multicast group 239.255.42.99...
[mocap_node-1] [INFO] [1657810175.838708537] [mocap_node]: Enabling non-blocking I/O
[mocap_node-1] [WARN] [1657810175.840185207] [mocap_node]: Client has not received server info request. Parsing data message aborted.
[mocap_node-1] [INFO] [1657810175.840958283] [mocap_node]: NATNet Version : 2.7.0.0
[mocap_node-1] [INFO] [1657810175.840974728] [mocap_node]: Server Version : 1.72.0.0
[mocap_node-1] [INFO] [1657810175.843955295] [mocap_node]: Initialization complete
[ERROR] [mocap_node-1]: process has died [pid 126850, exit code -11, cmd '/home/ros2/mocap_ws/install/mocap_optitrack/lib/mocap_optitrack/mocap_node --ros-args -r __node:=mocap_node --params-file /home/ros2/mocap_ws/install/mocap_optitrack/share/mocap_optitrack/config/mocap.yaml'].
```

The ROS 2 Galactic PC - IP: `192.168.5.100`

The Windows PC with Motive 1.7.2: IP: `192.168.5.40`, Motive had Multicast address: `192.168.5.100` (as per instructions - IP of PC with the ROS mocap_optitrack node), then I checked it with `239.255.42.99` too - I got the same error twice there.

The 

`mocap.yaml`:

```
mocap_node:
    ros__parameters:
        rigid_bodies:
            1:
                pose: "Robot_1/pose"
                pose2d: "Robot_1/ground_pose"
                child_frame_id: "Robot_1/base_link"
                parent_frame_id: "world"
            2:
                pose: "Robot_2/pose"
                pose2d: "Robot_2/ground_pose"
                child_frame_id: "Robot_2/base_link"
                parent_frame_id: "world"
            3:
                pose: "Robot_3/pose"
                pose2d: "Robot_3/ground_pose"
                child_frame_id: "Robot_3/base_link"
                parent_frame_id: "world"
            4:
                pose: "Robot_4/pose"
                pose2d: "Robot_4/ground_pose"
                child_frame_id: "Robot_4/base_link"
                parent_frame_id: "world"
            5:
                pose: "Robot_5/pose"
                pose2d: "Robot_5/ground_pose"
                child_frame_id: "Robot_5/base_link"
                parent_frame_id: "world"
        optitrack_config:
            multicast_address: "239.255.42.99"
            command_port: 1510
            data_port: 1511
```

How can I fix this? 

In the installed components I have:

```
NatNet Streaming Module (v. 2.7.0.0)
```

Perhaps this is the culprit?

shiven-verma, 2023-08-18:
Has this issue been solved? I am getting exact same error. My Natnet version is 2.10.0.0. I am using ROS2.

Berayksl, 2024-06-05:
I am also getting the same error

ljaniec, 2024-06-06:
Just FYI - we upgraded our Motive to 2.x version with new license and used [this MOCAP4ROS2 project](https://github.com/MOCAP4ROS2-Project/mocap4ros2_optitrack) instead

Berayksl, 2024-06-07:
Thank you for sharing this! How did you publish the mocap data to the appropriate MAVROS topic by using this package?

ljaniec, 2024-06-08:
I used it basically out-of-the-box, with modification of [launch file](https://github.com/MOCAP4ROS2-Project/mocap4ros2_optitrack/blob/rolling/mocap4r2_optitrack_driver/launch/optitrack2.launch.py)  and [the YAML file](https://github.com/MOCAP4ROS2-Project/mocap4ros2_optitrack/blob/rolling/mocap4r2_optitrack_driver/config/mocap4r2_optitrack_driver_params.yaml) to our Motive configuration, then I translated the rigid bodies to the odometry on the separate node + mapped a few topics, usual stuff and nothing extraordinary, I think.

Berayksl, 2024-06-08:
Which MAVROS topic did you publish your optitrack pose data to? I tried /local_position/pose but i kept gettin "VisOdom not healthy error"

ljaniec, 2024-06-09:
I didn't use MAVROS - the pose was used as an odometry source for Turtlebot2 mobile robot with Nav2.

closed 2024-06-09 by ljaniec